onitake/gsl-firmware #221: how to apply the settings on the onda W820 32gb tablet

manotroll, 2024-01-01:
https://github.com/onitake/gsl-firmware/tree/master/firmware/onda/v820w_32g

I copied the files to the lib/firmware folder
but he keeps complaining

[    0.249211] Spectre V2 : Enabling Restricted Speculation for firmware calls
[   16.211464] silead_ts i2c-MSSL1680:00: firmware: failed to load silead/gsl1680-onda-v820w-32g.fw (-2)
[   16.211559] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[   16.211655] silead_ts i2c-MSSL1680:00: firmware: failed to load silead/gsl1680-onda-v820w-32g.fw (-2)
[   16.211759] silead_ts i2c-MSSL1680:00: firmware: failed to load silead/gsl1680-onda-v820w-32g.fw (-2)
[   16.211854] silead_ts i2c-MSSL1680:00: firmware: failed to load silead/gsl1680-onda-v820w-32g.fw (-2)
[   16.211926] silead_ts i2c-MSSL1680:00: Direct firmware load for silead/gsl1680-onda-v820w-32g.fw failed with error -2

Could you have information on how to apply this in Debian 12?

jwrdegoede, 2024-01-01:
You need to create a silead folder under `/lib/firmware/` and then copy `gsl1680-onda-v820w-32g.fw` to `/lib/firmware/silead` that should do the trick.

manotroll, 2024-01-01:
but the file gsl1680-onda-v820w-32g.fw does not exist in the firmware/onda/v820w_32g directory