onitake/gsl-firmware #222: Touchscreen on Teclast x16 plus help (Maybe the same as Tbook11)

Phoenix-TW, 2024-01-09:
I was trying to install ubuntu on this tablet, everything works great except touchscreen. This tablet is a little bit special because it sell in Remix OS(Android) not with Windows version, but it can install Windows use teclast Tbook11's driver to activate touchcreen perfectly. X16 plus touchscreen resolution and size is the same with TBook11. I saw there is a TBook11 touchscreen driver in this repo , so I download and change name to mssl1680.fw then put it into /lib/firmware/siload. It can be load but it present abnormal. Every position is wrong. I am not familiar with firmware. Can anybody help me solve this problem?

ps. I have tried:
 - put siload_ts.fw in /lib/firmware
 - put gsl3692-teclast-tbook11.fw in /lib/firmware
 - put gsl3692-teclast-tbook11.fw /lib/firmware/siload
all of these can not load when system boot.
Only change gsl3692-teclast-tbook11.fw to mssl1680.fw and put in /lib/firmware/siload load successfully when system boot.

onitake, 2024-01-09:
That is interesting.

I suspect your problem is that the these devices may indeed have the same touchscreen configuration, but by simply renaming the firmware file to mssl1680.fw, you will end up with an incorrectly configured HID layer.
Sadly, we currently don't have any means to obtain device parameters from the firmware or the device itself, so we need to rely on kernel quirks to configure the driver. These quirks are specific to the device, not the touchscreen. If you try to use the same firmware on a different device that doesn't apply these quirks, the driver will use generic parameters that don't match the hardware.

If you're feeling a bit adventurous, you can modify the Tbook11 quirk so it also loads on the x16 plus:

Look at the instructions in the README section https://github.com/onitake/gsl-firmware/blob/master/README.md#silead_ts first - you're going to have to modify the Linux kernel sources and compile a new kernel.
In the `touchscreen_dmi_table` data structure, you'll find a matching entry for the Tbook11. Duplicate this entry and modify the `DMI_SYS_VENDOR`, `DMI_PRODUCT_NAME` and `DMI_PRODUCT_SKU` so they correspond with the x16 plus. You can find the correct values using `dmidecode`.

After recompiling and installing the modified kernel, you'll have to reboot. If you see the a dmesg line saying that it loaded `gsl3692-teclast-tbook11.fw`, you'll know that the override worked.

Please try this and report if this makes your touchscreen work correctly. We can then submit the changes for mainline kernel inclusion.

Phoenix-TW, 2024-01-10:
Thank you for your reply.
I use command dmidecode and found something like this:

```javascript=
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: TECLAST
	Product Name: Default string
	Version: Default string
	Serial Number: Default string
	UUID: 03000200-0400-0500-0006-000700080009
	Wake-up Type: Power Switch
	SKU Number: D3A5_A1
	Family: Default string

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: AMI Corporation
	Product Name: Cherry Trail CR
	Version: Default string
	Serial Number: Default string
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: Default string
	Type: Desktop
	Lock: Not Present
	Version: Default string
	Serial Number: Default string
	Asset Tag: Default string
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Default string
```
I want to check whether if these modify are correct because compile may cause much time.
SKU Number is Default string
I can't deside how to modify the code below.

```javascript=
static const struct property_entry teclast_x16plus_props[] = {
	PROPERTY_ENTRY_U32("touchscreen-min-x", 8),
	PROPERTY_ENTRY_U32("touchscreen-min-y", 14),
	PROPERTY_ENTRY_U32("touchscreen-size-x", 1916),
	PROPERTY_ENTRY_U32("touchscreen-size-y", 1264),
	PROPERTY_ENTRY_BOOL("touchscreen-inverted-y"),
	PROPERTY_ENTRY_STRING("firmware-name", "gsl3692-teclast-x16plus.fw"),
	PROPERTY_ENTRY_U32("silead,max-fingers", 10),
	PROPERTY_ENTRY_BOOL("silead,home-button"),
	{ }
};

static const struct ts_dmi_data teclast_(Default string) ??? _data = {
	.acpi_name	= "MSSL1680:00",
	.properties	= teclast_(Default string) ??? _props,
};

// Further below
{
	/* Teclast x16plus */
	.driver_data = (void *)&teclast_x16plus_data,
	.matches = {
		DMI_MATCH(DMI_SYS_VENDOR, "TECLAST"),
		DMI_MATCH(DMI_PRODUCT_NAME, "Default string"),
		DMI_MATCH(DMI_PRODUCT_SKU, "D3A5_A1"),
	},
},
```

 I also found there are some code in tbook11
".embedded_fw"
Are these code also need in x16plus?

Phoenix-TW, 2024-01-19:
silead_ts i2c-MSSL1680:00: Firmware request error -2

silead_ts i2c-MSSL1680:00: Initialization error, status: 0x0

Thank you for your help.
After a long struggle compiling kernel source, The silead_ts version work perfectly.

I added these code.
```c++
static const struct property_entry teclast_x16_plus_props[] = {
	PROPERTY_ENTRY_U32("touchscreen-min-x", 8),
	PROPERTY_ENTRY_U32("touchscreen-min-y", 14),
	PROPERTY_ENTRY_U32("touchscreen-size-x", 1916),
	PROPERTY_ENTRY_U32("touchscreen-size-y", 1264),
	PROPERTY_ENTRY_BOOL("touchscreen-inverted-y"),
	PROPERTY_ENTRY_STRING("firmware-name", "gsl3692-teclast-tbook11.fw"),
	PROPERTY_ENTRY_U32("silead,max-fingers", 10),
	PROPERTY_ENTRY_BOOL("silead,home-button"),
	{ }
};

static const struct ts_dmi_data teclast_x16_plus_data = {
	.embedded_fw = {
		.name	= "silead/gsl3692-teclast-tbook11.fw",
		.prefix = { 0xf0, 0x00, 0x00, 0x00, 0x02, 0x00, 0x00, 0x00 },
		.length	= 43560,
		.sha256	= { 0x9d, 0xb0, 0x3d, 0xf1, 0x00, 0x3c, 0xb5, 0x25,
			    0x62, 0x8a, 0xa0, 0x93, 0x4b, 0xe0, 0x4e, 0x75,
			    0xd1, 0x27, 0xb1, 0x65, 0x3c, 0xba, 0xa5, 0x0f,
			    0xcd, 0xb4, 0xbe, 0x00, 0xbb, 0xf6, 0x43, 0x29 },
	},
	.acpi_name	= "MSSL1680:00",
	.properties	= teclast_x16_plus_props,
}.

// Further below
	{
		/* Teclast X16 Plus */
		.driver_data = (void *)&teclast_x16_plus_data,
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "TECLAST"),
			DMI_MATCH(DMI_PRODUCT_NAME, "Default string"),
			DMI_MATCH(DMI_PRODUCT_SKU, "D3A5_A1"),
		},
	},
```

After installed new kernel, I put gsl3692-teclast-tbook11.fw into /lib/firmware/silead without renaming and reboot.
It finally worked.
I have uploaded a video on [Youtube](https://youtu.be/AO-UuIvuVwI).
I will submit a patch to patchwork later. The code will change gsl3692-teclast-tbook11.fw to gsl3692-teclast-x16-plus.fw

onitake, 2024-01-19:
That is awesome!
Thank you very much.

closed 2024-03-15 by Phoenix-TW