onitake/gsl-firmware #225: gsl2681 is not working properly

mauphuong, 2024-03-14:
Hello,
I am doing to add GSL2681 touch controller to raspberry pi. I already recomplied the kernel and added silead.ko module to system. The GSL2681 ID is 0xb4820000 and I am using the firmware: https://github.com/onitake/gsl-firmware/blob/master/firmware/linux/silead/gsl1680-b482-q8-d702.fw, but touch not working properly:
- Touch is A location but system reconigize B location. 
Anyone know how to fix and how to get firmware of GSL2681 IC?

thanks

mauphuong, 2024-03-14:
hi @onitake,
Thank for your quick reply. Actually, I have the device working on Android, and the touch controller working well. Now I want to reuse the screen with Raspberry PI. I dont have firmware image, so I get the exisiting image from your git. Do you know how to extract GSL controller from running Android?

thanks

onitake, 2024-03-14:
Take a look at the readme, it should help: https://github.com/onitake/gsl-firmware#android-driver

mauphuong, 2024-03-15:
Thank you, let me try but it seems to be impossible for my case