onitake/gsl-firmware #230: Extract firmware from the hardware

jpberdejo, 2024-04-07:
Hi.

I don't have the original OS for the Tibuta W120 tablet and I can´t find the specific firmware for this device anyware. ¿There is a way for extract or rebuild the firmware from the hardware?

barni2000, 2024-04-07:
Sometimes firmwares are packaged in SileadTouch.sys file. You could find the start of the firmware file with ```hexdump SileadTouch.sys | grep "00f0 0000 0002 0000 0000 0000 0000 0000"```

onitake, 2024-04-07:
Unfortunately not, because there _is_ no firmware in the hardware (except for a bootloader, maybe).
The firmware needs to be uploaded at runtime, which is what the driver does.

Yonits12, 2024-06-23:
When you say that the hardware does not contain any 'built in' firmware but (maybe) the bootloader, does it mean that the firmware-code which operates the controller's regular job (such as handle the conversion of the touch events sensed by the screen into the actual coordinates transferred over the i2c bus back to the driver, etc.) is entirely contained inside the runtime-uploaded firmware?

onitake, 2024-06-24:
@Yonits12 To my knowledge, this functionality is all contained in the firmware uploaded at runtime.
However, I am not the manufacturer and I cannot know for sure.