onitake/gsl-firmware #237: Mirrored half of the touchscreen along the X axis

KolhozniiSniper, 2024-07-03:
Hi, i have some several copies Jumper EzPad V10, and touchscreen drivers from the manufacturer for touchscreen.
On 3 copies everything was installed quite normally and works. But on 1 copy, with exactly the same touchscreen model, half of the panel is mirrored along the Y axis. 
I create with command:
> ../tools/fwtool -c firmware.fw -m 1680 -w 1525 -h 2035 -f track,swap,yflip -t 10 silead_ts.fw
Which works quite well on several copies.
It's the same problem, in the wrong touchscreen driver or something different?

olmonderald, 2024-10-08:
Hi, i am experiencing the same problem, if possible, upload the file with the correct driver

KolhozniiSniper, 2024-10-08:
> Hi, i am experiencing the same problem, if possible, upload the file with the correct driver

Unfortunately, I have not yet been able to achieve success with this task. I'm still working on it, but I'm running out of ideas.

htetoohlaing, 2024-12-05:
I guess I am in the same boat. My touch screen is responding in different places. Also I have tried the driver provided from the official Chinese website and still having same error. Any updates? (Sorry for my English)

AlexZonic100, 2024-12-07:
Hello, will this method works on windows OS? If not can you provide me your working copies plz. I really appreciate it.

KolhozniiSniper, 2025-02-27:
> I guess I am in the same boat. My touch screen is responding in different places. Also I have tried the driver provided from the official Chinese website and still having same error. Any updates? Also can you provide me WB_PADV10_3692_L_X64_230705_ZT.h file? The official website provide SileadTouch.sys for touch screen driver

[GDrive](https://drive.google.com/drive/folders/191xZNlrSbYC2hWNbCyg2q1Vyd4_wM8O6?usp=sharing)

Here I have attached all the drivers that I had from the manufacturer at the time of the last communication with him.  This is about the beginning of 2024.
I recently received a new EZpad V10, and noticed the label on it: "Model: JPG09". Previous models did not have this marking. I also noticed on the manufacturer's website that the touchscreen drivers were updated in September of the previous year. I haven't tested them at the moment.

[Official Site](https://www.jumper.com.cn/download/index.html)

This is a link to the official website with drivers, the EZpad V10 is only available in the Chinese version of the website.