onitake/gsl-firmware #248: is it possible to rebuild sileadtouch firmware since stock one is missing?

WeirdCoding123, 2025-02-26:
as title,my device dont have working firmware(wether android or windows)right now

onitake, 2025-02-26:
In theory, yes.

In practice, you'll either need to reverse engineer existing firmware (which is probably mostly parameter tables), or get your hands on the "Silead Ctouch Configure Suite", which is a proprietary piece of software for OEMs to configure/calibrate their touchscreens. Sadly, I don't have this software. I only know about it because it's mentioned in data sheets, such as [here](https://threefivedisplays.com/wp-content/uploads/datasheets/touch_driver_datasheets/GSL1688_REV_0_0_1_DS.pdf).

onitake, 2025-02-26:
The other option is to find someone with the same device and ask them for a dump of the Windows or Android driver...

WeirdCoding123, 2025-02-27:
> The other option is to find someone with the same device and ask them for a dump of the Windows or Android driver...

thanks for your reply
Sadly my device is a no brand tablet，model says“EM_H8811”.I am pretty sure its from company “emdoor”who product a load of crappy things and they never release their product software to consumer....
And this thing seems has different variant but identical model number，utilizes different touch controllers or same touch controller but different screen resolution,so others firmware might not work...Then I decided to figure it by myself

Thanks for your reply